themperek/cocotb-test #213: Possible race condition in plugin to mkdir ".cocotb-results" when using pytest-xdist

patrickrst, 2023-01-10:
The following error happened when running a testsuite with pytest-xdist: `FileExistsError: [Errno 17] File exists: '/project/.cocotb-results'`. This is a directory created in the pytest hook `pytest_sessionstart`, which by a quick search seems to be called by each worker (https://github.com/pytest-dev/pytest-xdist/issues/809). I haven't been able to reproduce it yet, but this seems like a possible race condition. Does that seems correct?

Log:
```
14:36:09  ============================= test session starts ==============================
14:36:09  platform linux -- Python 3.9.6, pytest-7.1.2, pluggy-1.0.0
14:36:09  rootdir: /project, configfile: pyproject.toml
14:36:09  plugins: cocotb-test-0.2.2, xdist-3.1.0
14:36:09  gw0 I / gw1 I / gw2 I / gw3 I / gw4 I / gw5 I / gw6 I / gw7 I / gw8 I / gw9 I / gw10 I / gw11 I / gw12 I / gw13 I / gw14 I / gw15 I
14:36:09  
[gw0] Python 3.9.6 (default, Aug 11 2021, 06:39:25)  -- [GCC 8.5.0 20210514 (Red Hat 8.5.0-3)]
14:36:09  
[gw1] Python 3.9.6 (default, Aug 11 2021, 06:39:25)  -- [GCC 8.5.0 20210514 (Red Hat 8.5.0-3)]
14:36:09  
[gw3] Python 3.9.6 (default, Aug 11 2021, 06:39:25)  -- [GCC 8.5.0 20210514 (Red Hat 8.5.0-3)]
14:36:09  
[gw4] Python 3.9.6 (default, Aug 11 2021, 06:39:25)  -- [GCC 8.5.0 20210514 (Red Hat 8.5.0-3)]
14:36:09  
[gw2] Python 3.9.6 (default, Aug 11 2021, 06:39:25)  -- [GCC 8.5.0 20210514 (Red Hat 8.5.0-3)]
14:36:09  
[gw6] Python 3.9.6 (default, Aug 11 2021, 06:39:25)  -- [GCC 8.5.0 20210514 (Red Hat 8.5.0-3)]
14:36:09  INTERNALERROR> def worker_internal_error(self, node, formatted_error):
14:36:09  INTERNALERROR>         """
14:36:09  INTERNALERROR>         pytest_internalerror() was called on the worker.
14:36:09  INTERNALERROR>     
14:36:09  INTERNALERROR>         pytest_internalerror() arguments are an excinfo and an excrepr, which can't
14:36:09  INTERNALERROR>         be serialized, so we go with a poor man's solution of raising an exception
14:36:09  INTERNALERROR>         here ourselves using the formatted message.
14:36:09  INTERNALERROR>         """
14:36:09  INTERNALERROR>         self._active_nodes.remove(node)
14:36:09  INTERNALERROR>         try:
14:36:09  INTERNALERROR> >           assert False, formatted_error
14:36:09  INTERNALERROR> E           AssertionError: Traceback (most recent call last):
14:36:09  INTERNALERROR> E               File "/usr/local/lib/python3.9/site-packages/_pytest/main.py", line 266, in wrap_session
14:36:09  INTERNALERROR> E                 config.hook.pytest_sessionstart(session=session)
14:36:09  INTERNALERROR> E               File "/usr/local/lib/python3.9/site-packages/pluggy/_hooks.py", line 265, in __call__
14:36:09  INTERNALERROR> E                 return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
14:36:09  INTERNALERROR> E               File "/usr/local/lib/python3.9/site-packages/pluggy/_manager.py", line 80, in _hookexec
14:36:09  INTERNALERROR> E                 return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
14:36:09  INTERNALERROR> E               File "/usr/local/lib/python3.9/site-packages/pluggy/_callers.py", line 60, in _multicall
14:36:09  INTERNALERROR> E                 return outcome.get_result()
14:36:09  INTERNALERROR> E               File "/usr/local/lib/python3.9/site-packages/pluggy/_result.py", line 60, in get_result
14:36:09  INTERNALERROR> E                 raise ex[1].with_traceback(ex[2])
14:36:09  INTERNALERROR> E               File "/usr/local/lib/python3.9/site-packages/pluggy/_callers.py", line 39, in _multicall
14:36:09  INTERNALERROR> E                 res = hook_impl.function(*args)
14:36:09  INTERNALERROR> E               File "/usr/local/lib/python3.9/site-packages/cocotb_test/plugin.py", line 24, in pytest_sessionstart
14:36:09  INTERNALERROR> E                 os.makedirs(self.results_xml_dir)
14:36:09  INTERNALERROR> E               File "/usr/lib64/python3.9/os.py", line 225, in makedirs
14:36:09  INTERNALERROR> E                 mkdir(name, mode)
14:36:09  INTERNALERROR> E             FileExistsError: [Errno 17] File exists: '/project/.cocotb-results'
14:36:09  INTERNALERROR> E           assert False
14:36:09  INTERNALERROR> 
14:36:09  INTERNALERROR> /usr/local/lib/python3.9/site-packages/xdist/dsession.py:190: AssertionError
14:36:09  [gw5] node down: Not properly terminated
14:36:09  
14:36:09  replacing crashed worker gw5
14:36:09  
[gw16] linux Python 3.9.6 cwd: /project
14:36:09  INTERNALERROR> Traceback (most recent call last):
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/_pytest/main.py", line 268, in wrap_session
14:36:09  INTERNALERROR>     session.exitstatus = doit(config, session) or 0
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/_pytest/main.py", line 322, in _main
14:36:09  INTERNALERROR>     config.hook.pytest_runtestloop(session=session)
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/pluggy/_hooks.py", line 265, in __call__
14:36:09  INTERNALERROR>     return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/pluggy/_manager.py", line 80, in _hookexec
14:36:09  INTERNALERROR>     return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/pluggy/_callers.py", line 60, in _multicall
14:36:09  INTERNALERROR>     return outcome.get_result()
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/pluggy/_result.py", line 60, in get_result
14:36:09  INTERNALERROR>     raise ex[1].with_traceback(ex[2])
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/pluggy/_callers.py", line 39, in _multicall
14:36:09  INTERNALERROR>     res = hook_impl.function(*args)
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/xdist/dsession.py", line 115, in pytest_runtestloop
14:36:09  INTERNALERROR>     self.loop_once()
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/xdist/dsession.py", line 138, in loop_once
14:36:09  INTERNALERROR>     call(**kwargs)
14:36:09  INTERNALERROR>   File "/usr/local/lib/python3.9/site-packages/xdist/dsession.py", line 228, in worker_errordown
14:36:09  INTERNALERROR>     self._active_nodes.remove(node)
14:36:09  INTERNALERROR> KeyError: <WorkerController gw5>
14:36:09  
14:36:09  ============================ no tests ran in 6.72s =============================
```

themperek, 2023-01-23:
Thank you for report. Should work better now.

closed 2023-01-23 by themperek

patrickrst, 2023-01-23:
Thank you for the fix!