themperek/cocotb-test #214: ERROR:          VHPI       NOTOTM (cocotb)

IgorAmosov, 2023-01-11:
```
Cannot iterate on type vhpiInternalRegions using a reference handle of type <unknown>. (cocotb)
         ../my_module.sv, 7: (cocotb)
```
 ../my_module.sv, 7:  - It is a definition of top module 
Hello! I am using cocotb==1.7.2 and cocotb-test==0.2.3 with Xcelium now.
When I run pure cocotb(through its Makefile) everything is OK!
But when i run it with cocotb-test i receive the error mentioned above. 
What does this error mean?

themperek, 2023-01-18:
Highly unlikely this is `cocotb-test`. There should te some other difference.

markusdd, 2025-02-28:
The issue is this: https://github.com/cocotb/cocotb/issues/4506

this is neither the fault of cocotb nor cocotb-test but an xcelium issue.

I have filed a PR with cocotb to work around the problem, an issue with Cadence is filed as well.

I think you can close this here as it is not the fault of cocotb-test

closed 2025-02-28 by themperek