themperek/cocotb-test #217: asyncio.streams.LimitOverrunError: Separator is not found, and chunk exceed the limit

alexforencich, 2023-01-19:
I spent all day debugging a hang that only seemed to be occurring when running under tox, which was quite annoying.  Finally managed to trace it to cocotb-test:

```
asyncio:base_events.py:1619 Task exception was never retrieved
future: <Task finished coro=<Simulator._log_pipe() done, defined at /home/alex/Projects/cocotbext-eth/.tox/py37/lib/python3.7/site-packages/cocotb_test/simulator.py:287> exception=ValueError('Separator is not found, and chunk exceed the limit')>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/streams.py", line 496, in readline
    line = await self.readuntil(sep)
  File "/usr/lib/python3.7/asyncio/streams.py", line 576, in readuntil
    offset)
asyncio.streams.LimitOverrunError: Separator is not found, and chunk exceed the limit

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/alex/Projects/cocotbext-eth/.tox/py37/lib/python3.7/site-packages/cocotb_test/simulator.py", line 289, in _log_pipe
    line = await stream.readline()
  File "/usr/lib/python3.7/asyncio/streams.py", line 505, in readline
    raise ValueError(e.args[0])
ValueError: Separator is not found, and chunk exceed the limit
```

Looks like long log lines trigger this exception, which is not handled by cocotb-test and as a result causes the whole shebang to hang.

Couple of things should be done here I think:

First, any exceptions when doing anything with the stdin/stdout streams should be handled in some way with try blocks.  This will prevent hangs like this going forward.

Second, we either need to specify a very high limit when opening the process (see https://stackoverflow.com/questions/55457370/how-to-avoid-valueerror-separator-is-not-found-and-chunk-exceed-the-limit), or avoid using readline/readuntil completely, or use readline/readuntil and fall back on some other method if there there is an `asyncio.streams.LimitOverrunError`.

In the meantime, I will roll back to an older version of cocotb-test on github actions.

themperek, 2023-01-22:
Is https://github.com/themperek/cocotb-test/pull/219 ok for now?

alexforencich, 2023-01-23:
Definitely seems to prevent the hang, yes.  But the log line in question is also truncated, so it's not a complete fix.  At least it's only one line though, not the whole log.

alexforencich, 2023-01-24:
Looks like the implementation of `readline()` in asyncio can actually serve as a template for properly handling arbitrarily long lines:

https://github.com/python/cpython/blob/3.11/Lib/asyncio/streams.py#L544-L554

PR #224 (2023-01-24): Support arbitrarily long log lines
This change should support arbitrarily long log lines from cocotb without truncating or hanging.  It's based on the implementation of `readline()` in asyncio, using `readuntil()` in combination with handling the appropriate exceptions.  Should fully fix #217.
closed 2023-01-24 by themperek