themperek/cocotb-test #229: cocotb-test not working when using Altera libraries

edian-annink, 2023-02-21:
Hi all, first of all thank you for writing this tool. It's a great idea and we would like to try it in our team. When using the following sim_arg `sim_args=['-L /home/edian/Applications/intelFPGA_lite/22.1std/questa_fse/intel/vhdl/altera_mf']` which works in the Makefile, the tool returns the following error: 
```
INFO     cocotb:simulator.py:334 Running command: vsim -c -onfinish exit -foreign cocotb_init /home/edian/.local/lib/python3.10/site-packages/cocotb/libs/libcocotbfli_modelsim.so -L /home/edian/Applications/intelFPGA_lite/22.1std/questa_fse/intel/vhdl/altera_mf project.project
INFO     cocotb:simulator.py:305 Reading pref.tcl
INFO     cocotb:simulator.py:305 
INFO     cocotb:simulator.py:305 # 2021.2
INFO     cocotb:simulator.py:305 
INFO     cocotb:simulator.py:305 # ** UI-Msg: (vish-3296) Unknown option '-L /home/edian/Applications/intelFPGA_lite/22.1std/questa_fse/intel/vhdl/altera_mf'.
INFO     cocotb:simulator.py:305 # Use the -help option for complete vsim usage.
ERROR    cocotb:simulator.py:305 Error loading design
```

 However, this option is present in the QuestaSim executable `vsim` as shown below:
```
# ------------------------------------Library------------------------------------
# Library Category for option
# 
# --------------------------------------------------------------------------------
# -L <libname>                    Search library for design units instantiated
#                                 from Verilog and for VHDL default component
#                                 binding

```
Is this a potential bug in cocotb-test or am I missing something?

themperek, 2023-02-21:
try:
`sim_args=['-L', '/home/edian/Applications/intelFPGA_lite/22.1std/questa_fse/intel/vhdl/altera_mf']`

edian-annink, 2023-02-22:
This works! Thank you so much :D

closed 2023-02-22 by edian-annink