themperek/cocotb-test #234: non-cocotb testbenches

vborchsh, 2023-05-18:
Did you think about mentioned feature? E.g.: we have mixed test environment. Partially in python+cocotb, the other part is pure system verilog.

So, it would be great to have an opportunity to run everything in one way - with cocotb-test. Because it makes easier switching between sim engines. But now it is quite complicated because of VPI and the other things needs to be provided to the simulator.

From my point it looks very easy: just skip any VPI, FLI... things in case of "native tb".

@themperek what do you think about it?

themperek, 2023-05-18:
This should be no problem. The only issue i see (at the moment) is you will not be able to know if simulation failed or not. As i remember (may be wrong) only some simulators give an exit code.

vborchsh, 2023-05-18:
This also an option. Thank you for response. Maybe, we'll come back with PRs.

closed 2023-05-18 by vborchsh