themperek/cocotb-test #236: Questa does not generate vsim.wlf when waves=True

Roenski, 2023-06-13:
Hi,

I'm trying to generate a waveform dump without opening the GUI by setting `waves=True`. For icarus and verilator this has worked fine, but questa does not seem to generate anything.

The problem seems to be with `self.do_script` defined in `simulator.py`. If `waves=True`, it is set as `log -recursive /*;`. However, at least on my configuration, this script does not generate a `vsim.wlf` file. However, if I set it as `log -recursive *;`, the file is generated. I.e. removing the forward slash seems to fix it.

I guess there is a slight chance that this is setup-specific? Maybe someone can test it on their machine?

themperek, 2023-06-23:
`log -recursive *;` works on my Modelsim.

Do you have some recent versions of Questa? 

I propose to change it to `log -recursive *;` Can you make PR?

One could also change this in `cocotb`  https://github.com/cocotb/cocotb/blob/65090a583040e4d19b4e433f6efeabd579a71f9a/cocotb/share/makefiles/simulators/Makefile.questa#L144C9-L144C26
Since cocotb has Questa in CI if one writes a test to check for this file it would check.

Roenski, 2023-06-27:
I have Questa 2017.01, so quite an old one. That's why I wanted to check if it's only my problem.

I can make a PR. Is there a way I can make the test and run it, since the CI is private?

themperek, 2023-06-27:
If you make a PR to `cocotb` it will run Questa. You cannot see it but maintainers do. 
But how to check that signals are in `wlf` file?

Roenski, 2023-06-27:
One test would be that the file gets generated at all.

Second would be to check that signals exist in the file. If the test module is known, it is probably enough to search for one of the signals lowest in the hierarchy. For that there is probably a `vsim` command, just need to browse through the command reference :)

themperek, 2023-06-30:
@Roenski 

Just found that `log -recursive *;` is not the same as `log -recursive /*;`. If I have multiple top design I see only the first one in waveforms in the case of `log -recursive *;`.

Roenski, 2023-08-15:
Hi, sorry for a bit late reply. I haven't done much to test this further. But, since you found that my suggestion breaks something, then, naturally, it shouldn't be used. Maybe it is just better to leave it as it is. It's not a big task to write custom code for generating the waveform, anyway :) 

I close the issue, if it is ok

closed 2023-08-15 by Roenski