themperek/cocotb-test #240: Incorrect cmd_build creation for VCS

alina-andreevna, 2023-07-23:
1. top level module name is not provided in cmd_build line. This is potential problem for simulation when user top module is not matched than found in compile list by vcs
2. timescale parameter syntax in vcs timescale=1ns/1ps, not timescale 1ns/1ps. Also it should be part of cmd_build, not common cmd

alina-andreevna, 2023-07-23:
I have a patch for this, but can not publish my branch:
```
ERROR: Permission to themperek/cocotb-test.git denied to alina-andreevna.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights     
and the repository exists.
```

themperek, 2023-08-05:
You need to fork this git repository and commit there and open pull request. Worst case province the code here.

themperek, 2023-08-11:
Close with https://github.com/themperek/cocotb-test/pull/244

closed 2023-08-11 by themperek