themperek/cocotb-test #241: for Questa/Modelsim is mixedsvvh flag necessary for verilog compilation?

adambagley, 2023-07-27:
Hi, just wondering if the mixedsvvh flag could potentially be removed from this line: https://github.com/themperek/cocotb-test/blob/4924adc78712434dcd076060777239c04a22bfe8/cocotb_test/simulator.py#L578

Seems maybe unnecessary when compiling verilog code? It ends up causing lots of spamming warning messages in the simulator log file.

Thank you,
Adam

manunited10, 2024-05-07:
Yeah gives error for me potentially for some complicated record of record type in a vhdl package (questasim)
```
** Error: (vcom-1609) The type used is not fully supported with the option -mixedsvvh.
```
Maybe we could make this `mixedsvvh ` optional?
It's not being generated when running cocotb through `makefile`.
Thanks.

adambagley, 2024-10-09:
Closing issue as it seems to have gotten fixed here https://github.com/themperek/cocotb-test/commit/1b06d12ee705d8007003b8299f89eddd5e8a10e4

closed 2024-10-09 by adambagley