themperek/cocotb-test #247: libcocotbfli_modelsim.so: wrong ELF class

jkwcht, 2023-09-25:
I am playing with a testbench that is running fine with a native cocotb makefile. It uses the Questa 2019.2 simulator. But running it from python using cocotb-test, I get an error from vsim: 

> The shared library /home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation.

The same simulator installation and python venv as the native cocotb call should be used. Do I miss any argument or configuration?

Here is the complete log:
```
-------------------------------------------------------------------------------------------- Captured log call ---------------------------------------------------------------------------------------------
INFO     cocotb:simulator.py:330 Running command: vlib spectrum_envelope
INFO     cocotb:simulator.py:302 ** Warning: (vlib-34) Library already exists at "spectrum_envelope".
INFO     cocotb:simulator.py:330 Running command: vcom -mixedsvvh -work spectrum_envelope -2008 /home/.../HW/src_hdl/config.vhd /home/.../HW/src_hdl/types_pkg.vhd /home/.../HW/src_hdl/spectrum_envelope.vhd
INFO     cocotb:simulator.py:302 QuestaSim vcom 2019.2 Compiler 2019.04 Apr 16 2019
INFO     cocotb:simulator.py:302 Start time: 14:57:47 on Sep 25,2023
INFO     cocotb:simulator.py:302 vcom -mixedsvvh -work spectrum_envelope -2008 /home/.../HW/src_hdl/config.vhd /home/.../HW/src_hdl/types_pkg.vhd /home/.../HW/src_hdl/spectrum_envelope.vhd
INFO     cocotb:simulator.py:302 -- Loading package STANDARD
INFO     cocotb:simulator.py:302 -- Loading package TEXTIO
INFO     cocotb:simulator.py:302 -- Loading package std_logic_1164
INFO     cocotb:simulator.py:302 -- Loading package NUMERIC_STD
INFO     cocotb:simulator.py:302 -- Compiling package config
INFO     cocotb:simulator.py:302 -- Loading package config
INFO     cocotb:simulator.py:302 -- Compiling package types_pkg
INFO     cocotb:simulator.py:302 -- Loading package types_pkg
INFO     cocotb:simulator.py:302 -- Compiling entity spectrum_envelope
INFO     cocotb:simulator.py:302 -- Compiling architecture rtl of spectrum_envelope
INFO     cocotb:simulator.py:302 End time: 14:57:47 on Sep 25,2023, Elapsed time: 0:00:00
INFO     cocotb:simulator.py:302 Errors: 0, Warnings: 0
INFO     cocotb:simulator.py:330 Running command: vsim -c -onfinish exit -foreign cocotb_init /home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so -voptargs=+acc -t 1ns spectrum_envelope.spectrum_envelope -do run -all; quit
INFO     cocotb:simulator.py:302 Reading pref.tcl
INFO     cocotb:simulator.py:302 
INFO     cocotb:simulator.py:302 # 2019.2
INFO     cocotb:simulator.py:302 
INFO     cocotb:simulator.py:302 # vsim -c -onfinish exit -foreign "cocotb_init /home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so" -voptargs="+acc" -t 1ns spectrum_envelope.spectrum_envelope -do "run -all; quit"
INFO     cocotb:simulator.py:302 # Start time: 14:57:48 on Sep 25,2023
INFO     cocotb:simulator.py:302 # ** Note: (vsim-8009) Loading existing optimized design _opt1
INFO     cocotb:simulator.py:302 # //  Questa Sim
INFO     cocotb:simulator.py:302 # //  Version 2019.2 linux Apr 16 2019
INFO     cocotb:simulator.py:302 # //
INFO     cocotb:simulator.py:302 # //  Copyright 1991-2019 Mentor Graphics Corporation
INFO     cocotb:simulator.py:302 # //  All Rights Reserved.
INFO     cocotb:simulator.py:302 # //
INFO     cocotb:simulator.py:302 # //  QuestaSim and its associated documentation contain trade
INFO     cocotb:simulator.py:302 # //  secrets and commercial or financial information that are the property of
INFO     cocotb:simulator.py:302 # //  Mentor Graphics Corporation and are privileged, confidential,
INFO     cocotb:simulator.py:302 # //  and exempt from disclosure under the Freedom of Information Act,
INFO     cocotb:simulator.py:302 # //  5 U.S.C. Section 552. Furthermore, this information
INFO     cocotb:simulator.py:302 # //  is prohibited from disclosure under the Trade Secrets Act,
INFO     cocotb:simulator.py:302 # //  18 U.S.C. Section 1905.
INFO     cocotb:simulator.py:302 # //
INFO     cocotb:simulator.py:302 # Loading std.standard
INFO     cocotb:simulator.py:302 # Loading std.textio(body)
INFO     cocotb:simulator.py:302 # Loading ieee.std_logic_1164(body)
INFO     cocotb:simulator.py:302 # Loading ieee.numeric_std(body)
INFO     cocotb:simulator.py:302 # Loading work.config
INFO     cocotb:simulator.py:302 # Loading work.types_pkg
INFO     cocotb:simulator.py:302 # Loading work.spectrum_envelope(rtl)#1
INFO     cocotb:simulator.py:302 # ** Warning: (vsim-3116) Problem reading symbols from /home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so : Not ELFCLASS32.
INFO     cocotb:simulator.py:302 # ** Warning: ** Warning: (vsim-3772) Failed to find the undefined symbols for shared library file "/home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so".
INFO     cocotb:simulator.py:302 #
INFO     cocotb:simulator.py:302 #    Time: 0 ns  Iteration: 0  Instance: /spectrum_envelope File: /home/.../HW/src_hdl/spectrum_envelope.vhd
INFO     cocotb:simulator.py:302 # Loading /home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so
INFO     cocotb:simulator.py:302 # ** Error: (vsim-125) The shared library /home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation. Please use compatible machines to build and load the library
INFO     cocotb:simulator.py:302 # ** Error (suppressible): (vsim-3197) Load of "/home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so" failed: /home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so: wrong ELF class: ELFCLASS64.
INFO     cocotb:simulator.py:302 # ** Warning: (vsim-FLI-3160) Failed to load FLI object file "/home/.../.venv/lib/python3.8/site-packages/cocotb/libs/libcocotbfli_modelsim.so".
INFO     cocotb:simulator.py:302 #    Time: 0 ns  Iteration: 0  Instance: /spectrum_envelope File: /home/.../HW/src_hdl/spectrum_envelope.vhd
INFO     cocotb:simulator.py:302 # run -all
INFO     cocotb:simulator.py:302 #  quit
INFO     cocotb:simulator.py:302 # End time: 14:57:48 on Sep 25,2023, Elapsed time: 0:00:00
INFO     cocotb:simulator.py:302 # Errors: 2, Warnings: 3
```

themperek, 2023-09-29:
Maybe add `-64` to vsim command?

jkwcht, 2023-10-05:
Thats it. Runs as expected after adding `sim_args=["-64"]`. Thanks!

closed 2023-10-05 by jkwcht