themperek/cocotb-test #255: `TypeError` exception when running simulation

cjmeyer, 2024-04-22:
When trying to run a simulation, a `TypeError` exception is being thrown; error shown below. I was finally able to track down the error to `libpython3.12.so` not being installed, but it would be nice if cocotb-test would be able to detect this case and provide a more informative error. I believe that cocotb-test could check if `find_libpython.find_libpython()` returned `None` or an empty string and raised an error at [simulator.py:207](https://github.com/themperek/cocotb-test/blob/d2e70f55a21573b9aa71caea2cc8322114867ba2/cocotb_test/simulator.py#L207) instead of setting `LIBRARY_LOC` to an empty string value which results in the `TypeError` getting thrown from `subprocess.py`.

cocotb version - 1.8, stable
cocotb-test version - 0.2.5 (installed w/ pip)
operating system - Ubuntu, 20.06.6 LTS, 64-bit
simulator version - GHDL, 4.0.0
python version - 3.12, in a virtual environment
error - 

```
/usr/lib/python3.12/asyncio/runners.py:194: in run
    return runner.run(main)
/usr/lib/python3.12/asyncio/runners.py:118: in run
    return self._loop.run_until_complete(task)
/usr/lib/python3.12/asyncio/base_events.py:685: in run_until_complete
    return future.result()
.venv/lib/python3.12/site-packages/cocotb_test/simulator.py:310: in _exec
    p = await asyncio.create_subprocess_exec(
/usr/lib/python3.12/asyncio/subprocess.py:224: in create_subprocess_exec
    transport, protocol = await loop.subprocess_exec(
/usr/lib/python3.12/asyncio/base_events.py:1742: in subprocess_exec
    transport = await self._make_subprocess_transport(
/usr/lib/python3.12/asyncio/unix_events.py:211: in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
/usr/lib/python3.12/asyncio/base_subprocess.py:36: in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
/usr/lib/python3.12/asyncio/unix_events.py:820: in _start
    self._proc = subprocess.Popen(
/usr/lib/python3.12/subprocess.py:1026: in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
/usr/lib/python3.12/subprocess.py:1870: in _execute_child
    env_list.append(k + b'=' + os.fsencode(v))
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

filename = None

>   ???
E   TypeError: expected str, bytes or os.PathLike object, not NoneType

<frozen os>:818: TypeError
```

constant007, 2024-12-02:
same err：
$ SIM=icarus pytest -o log_cli=True test_dff.py
Test session starts (platform: linux, Python 3.11.2, pytest 8.2.2, pytest-sugar 1.0.0)
rootdir: /home/enrsuwg/my_testbench/cocotb/dff
plugins: allure-pytest-2.13.5, metadata-3.1.1, html-4.1.1, cov-5.0.0, cocotb-test-0.2.5, xdist-3.6.1, sugar-1.0.0
collected 2 items

----------------------------------------------------------------------------- live log call ------------------------------------------------------------------------------
INFO     cocotb:simulator.py:334 Running command: iverilog -o /home/enrsuwg/my_testbench/cocotb/dff/sim_build/dff_test.vvp -D COCOTB_SIM=1 -g2012 -s dff_test -s iverilog_dump /home/enrsuwg/my_testbench/cocotb/dff/dff.sv /home/enrsuwg/my_testbench/cocotb/dff/sim_build/iverilog_dump.v

―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― test_dff_verilog ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

    @pytest.mark.skipif(os.getenv("SIM") in ("ghdl", "nvc"), reason="Verilog not suported")
    # @pytest.mark.parametrize("seed", range(10))
    def test_dff_verilog():
>       run(verilog_sources=[os.path.join(tests_dir, "dff.sv")], toplevel="dff_test", module="dff_cocotb", waves=True)  # sources  # top level HDL  # name of cocotb test module

test_dff.py:11:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
/usr/lib/python3.11/asyncio/runners.py:190: in run
    return runner.run(main)
/usr/lib/python3.11/asyncio/runners.py:118: in run
    return self._loop.run_until_complete(task)
/usr/lib/python3.11/asyncio/base_events.py:653: in run_until_complete
    return future.result()
/usr/local/lib/python3.11/dist-packages/cocotb_test/simulator.py:310: in _exec
    p = await asyncio.create_subprocess_exec(
/usr/lib/python3.11/asyncio/subprocess.py:218: in create_subprocess_exec
    transport, protocol = await loop.subprocess_exec(
/usr/lib/python3.11/asyncio/base_events.py:1694: in subprocess_exec
    transport = await self._make_subprocess_transport(
/usr/lib/python3.11/asyncio/unix_events.py:207: in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
/usr/lib/python3.11/asyncio/base_subprocess.py:36: in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
/usr/lib/python3.11/asyncio/unix_events.py:810: in _start
    self._proc = subprocess.Popen(
/usr/lib/python3.11/subprocess.py:1024: in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
/usr/lib/python3.11/subprocess.py:1821: in _execute_child
    env_list.append(k + b'=' + os.fsencode(v))
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

filename = None

>   ???
E   TypeError: expected str, bytes or os.PathLike object, not NoneType

<frozen os>:812: TypeError
--------------------------------------------------------------------------- Captured log call ----------------------------------------------------------------------------
INFO     cocotb:simulator.py:334 Running command: iverilog -o /home/enrsuwg/my_testbench/cocotb/dff/sim_build/dff_test.vvp -D COCOTB_SIM=1 -g2012 -s dff_test -s iverilog_dump /home/enrsuwg/my_testbench/cocotb/dff/dff.sv /home/enrsuwg/my_testbench/cocotb/dff/sim_build/iverilog_dump.v

 test_dff.py::test_dff_verilog ⨯                                                                                                                            50% █████
 test_dff.py::test_dff_vhdl s                                                                                                                              100% ██████████
======================================================================== short test summary info =========================================================================
FAILED test_dff.py::test_dff_verilog - TypeError: expected str, bytes or os.PathLike object, not NoneType

Results (0.33s):
       1 failed
         - test_dff.py:8 test_dff_verilog
       1 skipped

themperek, 2025-01-04:
Would be great if you could provide PR for this.