themperek/cocotb-test #278: cocotb 2.0.0 incompatibilites

bohrax, 2025-09-14:
When executing tests with pytest via cocotb-test in cocotb 2.0.0 there are two major differences between the behavior of cocotb 1.9.2:

1. Filtering tests with the `-k` flag does not work.
2. Multiple tests in the same python file runs in the same simulation session.

Example from cocotb 1.9.2:
```
pytest test_module.py -k first
=============================================================================================================== test session starts ================================================================================================================
platform linux -- Python 3.8.10, pytest-8.3.4, pluggy-1.5.0
rootdir: /home/bohre/workspace/fpga/core/cocotb
configfile: pytest.ini
plugins: cocotb-test-0.2.6
collected 2 items / 1 deselected / 1 selected                                                                                                                                                                                                      

test_module.py::test_module[first] 
------------------------------------------------------------------------------------------------------------------ live log call -------------------------------------------------------------------------------------------------------------------
19:02:27 INFO     cocotb               execute              Running command: ghdl -r --work=module module --vpi=/home/bohre/workspace/fpga/core/venv/python3/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so --wave=module.ghw
19:02:27 ERROR    cocotb               _log_pipe            loading VPI module '/home/bohre/workspace/fpga/core/venv/python3/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
19:02:27 INFO     cocotb               _log_pipe                 -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:108  in set_program_name_in_venv        Using Python virtual environment interpreter at /home/bohre/workspace/fpga/core/venv/python3/bin/python
19:02:27 INFO     cocotb               _log_pipe                 -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
19:02:27 ERROR    cocotb               _log_pipe            VPI module loaded!
19:02:27 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb                             Running on GHDL version 2.0.0-dev (1.0.0.r974.g0e46300c) [Dunoon edition]
19:02:27 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb                             Running tests with cocotb v1.9.2 from /home/bohre/workspace/fpga/core/venv/python3/lib/python3.8/site-packages/cocotb
19:02:27 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb                             Seeding Python random module with 1757869347
19:02:27 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb.regression                  Found test test_module.first
19:02:27 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb.regression                  running first (1/1)
19:02:27 INFO     cocotb               _log_pipe                25.00ns INFO     cocotb.regression                  first passed
19:02:27 INFO     cocotb               _log_pipe                25.00ns INFO     cocotb.regression                  **************************************************************************************
19:02:27 INFO     cocotb               _log_pipe                                                                    ** TEST                          STATUS  SIM TIME (ns)  REAL TIME (s)  RATIO (ns/s) **
19:02:27 INFO     cocotb               _log_pipe                                                                    **************************************************************************************
19:02:27 INFO     cocotb               _log_pipe                                                                    ** test_module.first              PASS          25.00           0.00      22121.86  **
19:02:27 INFO     cocotb               _log_pipe                                                                    **************************************************************************************
19:02:27 INFO     cocotb               _log_pipe                                                                    ** TESTS=1 PASS=1 FAIL=0 SKIP=0                 25.00           0.07        367.85  **
19:02:27 INFO     cocotb               _log_pipe                                                                    **************************************************************************************
19:02:27 INFO     cocotb               _log_pipe            
19:02:27 INFO     cocotb               run                  Results file: /home/bohre/workspace/fpga/core/cocotb/module_tb/sim_build/test_module/plve7xl0_results.xml
PASSED                                                                                                                                                                                                                                       [100%]

========================================================================================================= 1 passed, 1 deselected in 0.27s ==========================================================================================================
```
Example from 2.0.0:
```
pytest test_module.py  -k first
=============================================================================================================== test session starts ================================================================================================================
platform linux -- Python 3.8.10, pytest-8.3.5, pluggy-1.5.0
rootdir: /home/bohre/workspace/fpga/core/cocotb
configfile: pytest.ini
plugins: cocotb-test-0.2.6
collected 2 items / 1 deselected / 1 selected                                                                                                                                                                                                      

test_module.py::test_module[first] 
------------------------------------------------------------------------------------------------------------------ live log call -------------------------------------------------------------------------------------------------------------------
19:00:36 INFO     cocotb               execute              Running command: ghdl -r --work=module module --vpi=/home/bohre/workspace/fpga/core/venv2/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so --wave=module.ghw
19:00:36 ERROR    cocotb               _log_pipe            loading VPI module '/home/bohre/workspace/fpga/core/venv2/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_ghdl.so'
19:00:36 INFO     cocotb               _log_pipe                 -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:93   in _embed_init_python              Using Python 3.8.19 interpreter at /home/bohre/workspace/fpga/core/venv2/bin/python3
19:00:36 INFO     cocotb               _log_pipe                 -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:79   in gpi_print_registered_impl       VPI registered
19:00:36 ERROR    cocotb               _log_pipe            VPI module loaded!
19:00:36 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb                             Running on GHDL version 2.0.0-dev (1.0.0.r974.g0e46300c) [Dunoon edition]
19:00:36 INFO     cocotb               _log_pipe                 0.00ns WARNING  gpi                                vpi_iterate returned NULL for type vpiInstance for object NULL
19:00:36 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb                             Seeding Python random module with 1757869236
19:00:36 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb                             Initialized cocotb v2.0.0 from /home/bohre/workspace/fpga/core/venv2/lib/python3.8/site-packages/cocotb
19:00:37 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb                             Running tests
19:00:37 INFO     cocotb               _log_pipe                 0.00ns INFO     cocotb.regression                  running test_module.first (1/2)
19:00:37 INFO     cocotb               _log_pipe                15.00ns INFO     cocotb.regression                  test_module.first passed
[... cut ...]
19:00:37 INFO     cocotb               run                  Results file: /home/bohre/workspace/fpga/core/cocotb/module_tb/sim_build/test_module/w2etjrgo_results.xml
PASSED                                                                                                                                                                                                                                       [100%]

========================================================================================================= 1 passed, 1 deselected in 0.30s ==========================================================================================================

```

bohrax, 2025-09-17:
When replacing TESTCASE with COCOTB_TESTCASE I get a deprecation warning with the recommendation to use COCOTB_TEST_FILTER instead. As COCOTB_TEST_FILTER seems to work as intended, I'll go ahead and create a PR with that variable instead - if you don't have any objections?

themperek, 2025-09-17:
BTW: The way you use `conftest.py` is worth documenting somewhere. Maybe in the wiki https://github.com/cocotb/cocotb/wiki (probbly should use cocotb runner).

themperek, 2025-09-17:
Close #279